STMicroelectronics/meta-st-x-linux-ai #1: Tensorflow-lite-2.0.0-r0 do_compile ERROR

Igor197831100, 2020-01-24:
Hi,
   I am tring to compile the STM32MP1 AI but I have always the same error.
I have tried with Ubuntu 16.04 and 18.04 but the error is always the same.

These are the warning and error after launching the bitbake st-image-ai-cv:

`WARNING: tensorflow-lite-2.0.0-r0 do_fetch: Checksum mismatch for local file /home/user/STM32MP15-Ecosystem-v1.1.0/Distribution-Package/openstlinux-4.19-thud-mp1-19-10-09/build-openstlinuxweston-stm32mp1/downloads/v2.0.0.tar.gz
Cleaning and trying again.
WARNING: tensorflow-lite-2.0.0-r0 do_fetch: Renaming /home/user/STM32MP15-Ecosystem-v1.1.0/Distribution-Package/openstlinux-4.19-thud-mp1-19-10-09/build-openstlinuxweston-stm32mp1/downloads/v2.0.0.tar.gz to /home/user/STM32MP15-Ecosystem-v1.1.0/Distribution-Package/openstlinux-4.19-thud-mp1-19-10-09/build-openstlinuxweston-stm32mp1/downloads/v2.0.0.tar.gz_bad-checksum_5c711911d766d76813333c3812277574
WARNING: tensorflow-lite-2.0.0-r0 do_fetch: Checksum failure encountered with premirror download of https://github.com/tensorflow/tensorflow/archive/v2.0.0.tar.gz - will attempt other sources.
ERROR: tensorflow-lite-2.0.0-r0 do_compile: Function failed: do_compile (log file is located at /home/user/STM32MP15-Ecosystem-v1.1.0/Distribution-Package/openstlinux-4.19-thud-mp1-19-10-09/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-openstlinux_weston-linux-gnueabi/tensorflow-lite/2.0.0-r0/temp/log.do_compile.62803)
ERROR: Logfile of failure stored in: /home/user/STM32MP15-Ecosystem-v1.1.0/Distribution-Package/openstlinux-4.19-thud-mp1-19-10-09/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-openstlinux_weston-linux-gnueabi/tensorflow-lite/2.0.0-r0/temp/log.do_compile.62803
`

vinceab, 2020-01-27:
Hi,

This issue is due to the fact that 2 downloaded packages have exactly the same name v2.0.0.tar.gz but with different content.
The yocto build system is downloading a first time a package called v2.0.0.tar.gz for a package xxx and when it is time to build TensorFlowLite, the same package name is downloaded for TensorFlowLite. But instead of downloading it directly from github it looks first in the downloads directory and found the good package name (v2.0.0.tar.gz) but with the wrong checksum. As a consequence, the build is failing because the proper TensorFlowLite package is not  downloaded.

Can you please test the attached file and give me some feedbacks?
cd /ayers/meta-st/meta-st-stm32mpu-ai
git am 0001-tensorflow-lite-fix-fetch-issue.patch

[0001-tensorflow-lite-fix-fetch-issue.zip](https://github.com/STMicroelectronics/meta-st-stm32mpu-ai/files/4116298/0001-tensorflow-lite-fix-fetch-issue.zip)

vinceab, 2020-01-27:
Hi 

Patch tested successfully and now added in the master and thud branch.

BR
Vincent

commit 2b2959bd (2020-01-27): tensorflow-lite: fix fetch issue

prefix the downloaded package with tensorflow-lite prefix to avoid any
duplicated name of downloaded package.
It avoid the build system to fail while the fetch of tensorflow-lite
package.

Closes #1
closed 2020-01-27 by vinceab