STMicroelectronics/meta-st-x-linux-ai #2: ERROR: tensorflow-lite-2.0.0-r0 do_compile

Igor197831100, 2020-01-30:
Hi, I still have an issue during compiling:

`ERROR: tensorflow-lite-2.0.0-r0 do_compile: Function failed: do_compile (log file is located at /home/user/STM32MP15-Ecosystem-v1.1.0/Distribution-Package/openstlinux-4.19-thud-mp1-19-10-09/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-openstlinux_weston-linux-gnueabi/tensorflow-lite/2.0.0-r0/temp/log.do_compile.61543)
ERROR: Logfile of failure stored in: /home/user/STM32MP15-Ecosystem-v1.1.0/Distribution-Package/openstlinux-4.19-thud-mp1-19-10-09/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-openstlinux_weston-linux-gnueabi/tensorflow-lite/2.0.0-r0/temp/log.do_compile.61543
Log data follows:
| DEBUG: Executing shell function do_compile
| /home/user/STM32MP15-Ecosystem-v1.1.0/Distribution-Package/openstlinux-4.19-thud-mp1-19-10-09/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-openstlinux_weston-linux-gnueabi/tensorflow-lite/2.0.0-r0/tensorflow-2.0.0/tensorflow/lite/tools/pip_package/build_pip_package.sh: 22: /home/user/STM32MP15-Ecosystem-v1.1.0/Distribution-Package/openstlinux-4.19-thud-mp1-19-10-09/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-openstlinux_weston-linux-gnueabi/tensorflow-lite/2.0.0-r0/tensorflow-2.0.0/tensorflow/lite/tools/pip_package/build_pip_package.sh: Bad substitution
| grep: /home/user/STM32MP15-Ecosystem-v1.1.0/Distribution-Package/openstlinux-4.19-thud-mp1-19-10-09/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-openstlinux_weston-linux-gnueabi/tensorflow-lite/2.0.0-r0/tensorflow-2.0.0/../../../../tensorflow/tools/pip_package/setup.py: No such file or directory
| /home/user/STM32MP15-Ecosystem-v1.1.0/Distribution-Package/openstlinux-4.19-thud-mp1-19-10-09/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-openstlinux_weston-linux-gnueabi/tensorflow-lite/2.0.0-r0/tensorflow-2.0.0/tensorflow/lite/tools/pip_package/build_pip_package.sh: 30: /home/user/STM32MP15-Ecosystem-v1.1.0/Distribution-Package/openstlinux-4.19-thud-mp1-19-10-09/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-openstlinux_weston-linux-gnueabi/tensorflow-lite/2.0.0-r0/tensorflow-2.0.0/tensorflow/lite/tools/pip_package/build_pip_package.sh: [[: not found
| cp: cannot stat '/home/user/STM32MP15-Ecosystem-v1.1.0/Distribution-Package/openstlinux-4.19-thud-mp1-19-10-09/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-openstlinux_weston-linux-gnueabi/tensorflow-lite/2.0.0-r0/tensorflow-2.0.0/../../../../tensorflow/lite/python/interpreter_wrapper': No such file or directory
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_compile (log file is located at /home/user/STM32MP15-Ecosystem-v1.1.0/Distribution-Package/openstlinux-4.19-thud-mp1-19-10-09/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-openstlinux_weston-linux-gnueabi/tensorflow-lite/2.0.0-r0/temp/log.do_compile.61543)
ERROR: Task (/home/user/STM32MP15-Ecosystem-v1.1.0/Distribution-Package/openstlinux-4.19-thud-mp1-19-10-09/layers/meta-st/meta-st-stm32mpu-ai/recipes-frameworks/tensorflow-lite/tensorflow-lite_2.0.0.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 5373 tasks of which 5369 didn't need to be rerun and 1 failed.
NOTE: Writing buildhistory
`

Thank you.

vinceab, 2020-01-30:
The issue is due to the fact that the shell  point to dash and not bash.
The script /home/user/STM32MP15-Ecosystem-v1.1.0/Distribution-Package/openstlinux-4.19-thud-mp1-19-10-09/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-openstlinux_weston-linux-gnueabi/tensorflow-lite/2.0.0-r0/tensorflow-2.0.0/tensorflow/lite/tools/pip_package/build_pip_package.sh should be executed in a bash shell.

Can you please test it?

BR
Vincent

vinceab, 2020-01-30:
In parallel, I am preparing a patch to fix this issue.

Igor197831100, 2020-01-30:
echo $0 says my shell is a bash. Is it ok?

commit 1a219ba4 (2020-02-06): tensorflow-lite: fix compile issue due to bashisms in TFLite build script

Add a patch to remove bashisms in the TFLite build script build_pip_package.sh.

Closes #2
closed 2020-02-06 by vinceab