STMicroelectronics/meta-st-x-linux-ai #3: ERROR: tensorflow-lite-cv-apps-1.0-r0 do_compile: oe_runmake failed

kirillovk, 2020-02-11:
The previous error "ERROR: tensorflow-lite-2.0.0-r0 do_compile" has disappeared and this stage has passed perfectly

There was a new error during compilation

"ERROR: tensorflow-lite-cv-apps-1.0-r0 do_compile: oe_runmake failed
ERROR: tensorflow-lite-cv-apps-1.0-r0 do_compile: Function failed: do_compile "

detailed log is attached to the file "log.txt"

What could be the reason?
[log.txt](https://github.com/STMicroelectronics/meta-st-stm32mpu-ai/files/4184044/log.txt)

vinceab, 2020-02-11:
Hi,

After having a look to the log.txt, the build went wrong during the the link process because the wrapper_tfl.o file is truncated (wrapper_tfl.o: file not recognized: file truncated).
This issue is not due to the build process but more likely to your build environment (maybe a lack of memory).

Vincent

kirillovk, 2020-02-12:
Hello.

I restarted the assembly in the virtual machine.
At the same time on the host machine, I watched the use of RAM. During the assembly of large objects, the volume of the used memory of the host machine increased. The previous assembly failed with the reason that during the build on the virtual machine, other applications using a large amount of RAM were launched on the host machine. When running only a virtual machine, everything went without errors.

Thanks Vincent.

closed 2020-02-12 by kirillovk