STMicroelectronics/meta-st-x-linux-ai #4: Unable to install AI packages on STM32MP157A-DK1

jwyune, 2020-07-22:
I am trying to install AI packages on STM32MP157A-DK1. I followed the instruction from [here](https://wiki.st.com/stm32mpu/wiki/X-LINUX-AI_OpenSTLinux_Expansion_Package#Install_AI_packages).

The board was initialized with "FlashLayout_sdcard_stm32mp157a-dk1-trusted.tsv". The instruction is described [here](https://wiki.st.com/stm32mpu/wiki/Getting_started/STM32MP1_boards/STM32MP157x-DK2/Let%27s_start/Populate_the_target_and_boot_the_image).

However, I get the following error.

```
root@stm32mp1:~# apt-get install packagegroup-x-linux-ai
Reading package lists... Done
Building dependency tree
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 packagegroup-x-linux-ai : Depends: packagegroup-x-linux-ai-armnn-tflite but it is not going to be installed
                           Depends: packagegroup-x-linux-ai-tflite but it is not going to be installed
                           Depends: packagegroup-x-linux-ai-tflite-edgetpu but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
```

What is missing?

vinceab, 2020-07-22:
@jwyune can you please confirm that you are using either OpenSTLinux v2.0.0 or OpenSTLinux v1.2.0.
Further what is the board you are using STM32MP157a-DK1 or STM32MP157C-DK2?

Vincent

jwyune, 2020-07-22:
@vinceab 

1. I think I am using v1.2.0. The name of the tar.xz file is "en.FLASH-stm32mp1-openstlinux-20-02-19.tar.xz".
2. I am using STM32MP157A-DK1.

It seems like I should upgrade the version to v2.0.0.

vinceab, 2020-07-22:
@jwyune as you are using OpenSTLinux v1.2.0, either you update it to OpenSTLinux 2.0.0 or you can follow this wiki article that explain how to install X-LINUX-AI v2.0.0 on OpenSTLinux v1.2.0:
https://wiki.st.com/stm32mpu/wiki/How_to_install_X-LINUX-AI_v2.0.0_on_OpenSTLinux_v1.2.0

Vincent

jwyune, 2020-07-22:
@vinceab 

Thanks for your advice. That actually worked. I'm slowly getting used to the STM32MP1 documentations.

By the way, if I try to flash v2.0.0, I get this error on serial terminal: `dfu_write: Wrong sequence number! [1] [0]` Because of this error, I am unable to flash the latest Linux version on DK1. Where should I put this issue?

vinceab, 2020-07-22:
@jwyune regarding the flashing issue for the v2.0.0, it is better to have a look to the ST community website https://community.st.com/s/ where you can ask for question.

I close the issue.
Vincent

closed 2020-07-22 by vinceab