STMicroelectronics/meta-st-x-linux-ai #5: ERROR: tf-a-stm32mp-2.2.r1-r0 do_compile: oe_runmake failed

kirillovk, 2020-08-07:
I tried to compile st-image-ah.
Got the following error
------------------------------------------------------------------------------
Loaded 3827 entries from dependency cache.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.46.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "arm-ostl-linux-gnueabi"
MACHINE              = "stm32mp1"
DISTRO               = "openstlinux-weston"
DISTRO_VERSION       = "3.1-snapshot-20200807"
TUNE_FEATURES        = "arm vfp cortexa7 neon vfpv4 thumb callconvention-hard"
TARGET_FPU           = "hard"
DISTRO_CODENAME      = "dunfell"
ACCEPT_EULA_stm32mp1 = "1"
GCCVERSION           = "9.%"
PREFERRED_PROVIDER_virtual/kernel = "linux-stm32mp"
meta-python          
meta-oe              
meta-gnome           
meta-xfce            
meta-initramfs       
meta-multimedia      
meta-networking      
meta-webserver       
meta-filesystems     
meta-perl            = "HEAD:b5f510e48080b6dc710ff4800feb90ef679c5456"
meta-st-stm32mp      = "HEAD:7a1ab80ccb655e9cd28c9ea8a4a83fca48c48b11"
meta-qt5             = "HEAD:ac7514e2bc295c97faa86d24e8823829257adbb8"
meta-st-openstlinux  = "HEAD:5465ed3781355776cb46d2cc45884b1ccc1011cd"
meta                 = "HEAD:1795f30d8ab73d35710ca99064c51190dc84853e"
meta-st-stm32mpu-ai  = "HEAD:f7da03d6bb5e5285d75d7871d55faacf9aed485a"

Initialising tasks: 100% |#####################################################################################################| Time: 0:00:13
Sstate summary: Wanted 1580 Found 552 Missed 1028 Current 2000 (34% match, 71% complete)
NOTE: Executing Tasks
ERROR: tf-a-stm32mp-2.2.r1-r0 do_compile: oe_runmake failed
ERROR: tf-a-stm32mp-2.2.r1-r0 do_compile: Execution of '/home/osboxes/STM32MPU_workspace/Distribution-Package/openstlinux-5.4-dunfell-mp1-20-06-24/build-openstlinuxweston-stm32mp1/tmp-glibc/work/stm32mp1-ostl-linux-gnueabi/tf-a-stm32mp/2.2.r1-r0/temp/run.do_compile.78371' failed with exit code 1:
make: Entering directory '/home/osboxes/STM32MPU_workspace/Distribution-Package/openstlinux-5.4-dunfell-mp1-20-06-24/build-openstlinuxweston-stm32mp1/tmp-glibc/work-shared/stm32mp1/tfa-source'
make: *** No targets specified and no makefile found.  Stop.
make: Leaving directory '/home/osboxes/STM32MPU_workspace/Distribution-Package/openstlinux-5.4-dunfell-mp1-20-06-24/build-openstlinuxweston-stm32mp1/tmp-glibc/work-shared/stm32mp1/tfa-source'
WARNING: exit code 1 from a shell command.

ERROR: Logfile of failure stored in: /home/osboxes/STM32MPU_workspace/Distribution-Package/openstlinux-5.4-dunfell-mp1-20-06-24/build-openstlinuxweston-stm32mp1/tmp-glibc/work/stm32mp1-ostl-linux-gnueabi/tf-a-stm32mp/2.2.r1-r0/temp/log.do_compile.78371
Log data follows:
| DEBUG: Executing python function tfaconfig_env
| DEBUG: Python function tfaconfig_env finished
| DEBUG: Executing shell function do_compile
| NOTE: make -j 3 CROSS_COMPILE=arm-ostl-linux-gnueabi- DEBUG=1 LOG_LEVEL=40 PLAT=stm32mp1 ARCH=aarch32 ARM_ARCH_MAJOR=7 STM32MP_SDMMC=1 STM32MP_EMMC=1 STM32MP_SPI_NOR=1 STM32MP_RAW_NAND=1 STM32MP_SPI_NAND=1 -C /home/osboxes/STM32MPU_workspace/Distribution-Package/openstlinux-5.4-dunfell-mp1-20-06-24/build-openstlinuxweston-stm32mp1/tmp-glibc/work/stm32mp1-ostl-linux-gnueabi/tf-a-stm32mp/2.2.r1-r0/git DTB_FILE_NAME=stm32mp157a-dk1.dtb BUILD_PLAT=/home/osboxes/STM32MPU_workspace/Distribution-Package/openstlinux-5.4-dunfell-mp1-20-06-24/build-openstlinuxweston-stm32mp1/tmp-glibc/work/stm32mp1-ostl-linux-gnueabi/tf-a-stm32mp/2.2.r1-r0/build/trusted AARCH32_SP=sp_min
| make: Entering directory '/home/osboxes/STM32MPU_workspace/Distribution-Package/openstlinux-5.4-dunfell-mp1-20-06-24/build-openstlinuxweston-stm32mp1/tmp-glibc/work-shared/stm32mp1/tfa-source'
| make: *** No targets specified and no makefile found.  Stop.
| make: Leaving directory '/home/osboxes/STM32MPU_workspace/Distribution-Package/openstlinux-5.4-dunfell-mp1-20-06-24/build-openstlinuxweston-stm32mp1/tmp-glibc/work-shared/stm32mp1/tfa-source'
| ERROR: oe_runmake failed
| WARNING: exit code 1 from a shell command.
| ERROR: Execution of '/home/osboxes/STM32MPU_workspace/Distribution-Package/openstlinux-5.4-dunfell-mp1-20-06-24/build-openstlinuxweston-stm32mp1/tmp-glibc/work/stm32mp1-ostl-linux-gnueabi/tf-a-stm32mp/2.2.r1-r0/temp/run.do_compile.78371' failed with exit code 1:
| make: Entering directory '/home/osboxes/STM32MPU_workspace/Distribution-Package/openstlinux-5.4-dunfell-mp1-20-06-24/build-openstlinuxweston-stm32mp1/tmp-glibc/work-shared/stm32mp1/tfa-source'
| make: *** No targets specified and no makefile found.  Stop.
| make: Leaving directory '/home/osboxes/STM32MPU_workspace/Distribution-Package/openstlinux-5.4-dunfell-mp1-20-06-24/build-openstlinuxweston-stm32mp1/tmp-glibc/work-shared/stm32mp1/tfa-source'
| WARNING: exit code 1 from a shell command.
| 
ERROR: Task (/home/osboxes/STM32MPU_workspace/Distribution-Package/openstlinux-5.4-dunfell-mp1-20-06-24/layers/meta-st/meta-st-stm32mp/recipes-bsp/trusted-firmware-a/tf-a-stm32mp_2.2.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 7197 tasks of which 6821 didn't need to be rerun and 1 failed.
NOTE: Writing buildhistory
NOTE: Writing buildhistory took: 8 seconds

Summary: 1 task failed:
  /home/osboxes/STM32MPU_workspace/Distribution-Package/openstlinux-5.4-dunfell-mp1-20-06-24/layers/meta-st/meta-st-stm32mp/recipes-bsp/trusted-firmware-a/tf-a-stm32mp_2.2.bb:do_compile
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
---------------------------------------------------------------------------

My system environment
virtual machine
ubuntu 18.04 osboxes
CPU 3 core 
Memory 6 Gbyte 
STM32MP1 Distribution Package v2.0.0
X-LINUX-AI OpenSTLinux Expansion Package Version v2.0.0

vinceab, 2020-08-07:
@kirillovk tf-a-stm32mp is not part of the X-LINUX-AI expansion package. So it seems that you have an issue to build the STM32MP1 Distribution Package v2.0.0. Can you please try to build the st-image-weston first?

kirillovk, 2020-08-08:
@vinceab
I build STM32MP1 Distribution Package v2.0.0 st-image-weston image.

bitbake st-image-weston
NOTE: Started PRServer with DBfile: /home/osboxes/STM32MPU_workspace/Distribution-Package/openstlinux-5.4-dunfell-mp1-20-06-24/build-openstlinuxweston-stm32mp1/cache/prserv.sqlite3, IP: 127.0.0.1, PORT: 39581, PID: 3014
NOTE: /home/osboxes/STM32MPU_workspace/Distribution-Package/openstlinux-5.4-dunfell-mp1-20-06-24/layers/meta-st/meta-st-openstlinux/recipes-st/images/st-image-resize-initrd.bb: ENABLE_PARTITIONS_IMAGE not enabled
Parsing recipes: 100% |########################################################################################################| Time: 0:04:05
Parsing of 2614 .bb files complete (0 cached, 2614 parsed). 3805 targets, 106 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.46.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-18.04"
TARGET_SYS           = "arm-ostl-linux-gnueabi"
MACHINE              = "stm32mp1"
DISTRO               = "openstlinux-weston"
DISTRO_VERSION       = "3.1-snapshot-20200807"
TUNE_FEATURES        = "arm vfp cortexa7 neon vfpv4 thumb callconvention-hard"
TARGET_FPU           = "hard"
DISTRO_CODENAME      = "dunfell"
ACCEPT_EULA_stm32mp1 = "1"
GCCVERSION           = "9.%"
PREFERRED_PROVIDER_virtual/kernel = "linux-stm32mp"
meta-python          
meta-oe              
meta-gnome           
meta-xfce            
meta-initramfs       
meta-multimedia      
meta-networking      
meta-webserver       
meta-filesystems     
meta-perl            = "HEAD:b5f510e48080b6dc710ff4800feb90ef679c5456"
meta-st-stm32mp      = "HEAD:7a1ab80ccb655e9cd28c9ea8a4a83fca48c48b11"
meta-qt5             = "HEAD:ac7514e2bc295c97faa86d24e8823829257adbb8"
meta-st-openstlinux  = "HEAD:5465ed3781355776cb46d2cc45884b1ccc1011cd"
meta                 = "HEAD:1795f30d8ab73d35710ca99064c51190dc84853e"

NOTE: Fetching uninative binary shim from http://downloads.yoctoproject.org/releases/uninative/2.8/x86_64-nativesdk-libc.tar.xz;sha256sum=a09922172c3a439105e0ae6b943daad2d83505b17da0aba97961ff433b8c21ab
Initialising tasks: 100% |#####################################################################################################| Time: 0:00:22
Sstate summary: Wanted 3305 Found 0 Missed 3305 Current 0 (0% match, 0% complete)
NOTE: Executing Tasks
WARNING: st-initrd-1.0-r0 do_package: Manifest /home/osboxes/STM32MPU_workspace/Distribution-Package/openstlinux-5.4-dunfell-mp1-20-06-24/build-openstlinuxweston-stm32mp1/tmp-glibc/sstate-control/manifest-x86_64_x86_64-nativesdk-st-image-resize-initrd.packagedata not found in stm32mp1 cortexa7t2hf-neon-vfpv4 cortexa7t2hf-neon cortexa7t2hf-vfp cortexa7hf-neon-vfpv4 cortexa7hf-neon cortexa7hf-vfp armv7vet2hf-neon-vfpv4 armv7vehf-neon-vfpv4 armv7vet2hf-neon armv7vehf-neon armv7vet2hf-vfp armv7vehf-vfp armv7at2hf-vfp armv7ahf-vfp armv6thf-vfp armv6hf-vfp armv5tehf-vfp armv5ehf-vfp armv5thf-vfp armv5hf-vfp allarch x86_64_x86_64-nativesdk (variant '')?
NOTE: Tasks Summary: Attempted 9058 tasks of which 5 didn't need to be rerun and all succeeded.
NOTE: Writing buildhistory
NOTE: Writing buildhistory took: 47 seconds

Summary: There was 1 WARNING message shown.

Or would you like to say that I would first build st-image-weston and then build X-LINUX-AI?

vinceab, 2020-08-10:
@kirillovk building st-image-weston allows you to be sure the tf-a package is building properly. It seems it is working. 
building st-image-ai should not differ from st-image-weston for tf-a.
I will check this issue when I will back to office.
Vincent

vinceab, 2020-10-19:
Hello @kirillovk,

Are you still facing build issue?
I made multiple test with different environment and I did not manage to reproduce this issue.

BR
Vincent

vinceab, 2021-02-08:
Issue not seen anymore.
Close it

closed 2021-02-08 by vinceab