STMicroelectronics/meta-st-x-linux-ai #6: Including PyCoral and LibCoral modules in the X-LINUX-AI OpenSTLinux Expansion Package

ramkumarkoppu, 2021-06-13:
Hi, To use features such as on-device transfer learning using Google Coral TPU, we would like to have Google's Pycoral and libcoral modules in the  X-LINUX-AI OpenSTLinux Expansion Package. Using standard TF lite library provided with X-LINUX-AI OpenSTLinux Expansion Package would not support these advanced features atleast with currently it stands. Can you please distribute these lbraries in this package for STM32MP1 devices? We thought of building these libraries from source but it relies on bazel build system and we don't know how to use it with ST's OpenSTLinux.

vinceab, 2021-06-14:
@ramkumarkoppu,

The support of Pycoral and libcoral is planned but as of today, I cannot provide any release date.
The way pycoral and libcoral can be build using Open Embedded is the same way I did for libedgetpu build:
https://github.com/STMicroelectronics/meta-st-stm32mpu-ai/tree/master/recipes-frameworks/libedgetpu

A specific make file need to be added to the libcoral ou pycoral in order to create the Openembedded recipe.
It is a handmade translation from bazel to Makefile.

Vincent

vinceab, 2021-11-26:
Issue close since no update since June 14 2021

closed 2021-11-26 by vinceab