STMicroelectronics/meta-st-x-linux-ai #9: ERROR : Onnxruntime_git.bb:do_configure

IlLupo12725, 2024-11-15:
lupo@lupo:~/PhytechSTM/build-openstlinuxweston-phycore-stm32mp1-3$ bitbake st-image-weston
NOTE: Started PRServer with DBfile: /home/lupo/PhytechSTM/build-openstlinuxweston-phycore-stm32mp1-3/cache/prserv.sqlite3, Address: 127.0.0.1:45403, PID: 317946
Loading cache: 100% |#####################################################################################################################| Time: 0:00:00
Loaded 4557 entries from dependency cache.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "2.0.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "arm-ostl-linux-gnueabi"
MACHINE              = "phycore-stm32mp1-3"
DISTRO               = "openstlinux-weston"
DISTRO_VERSION       = "4.0.19-BSP-Yocto-OpenSTLinux-STM32MP1-PD23.1.1"
TUNE_FEATURES        = "arm vfp cortexa7 neon vfpv4 thumb callconvention-hard"
TARGET_FPU           = "hard"
DISTRO_CODENAME      = "kirkstone"
ACCEPT_EULA_phycore-stm32mp1-3 = "1"
GCCVERSION           = "11.%"
PREFERRED_PROVIDER_virtual/kernel = "linux-stm32mp"
meta-phytec          = "HEAD:81e913ee53aeb47d48b426683dc2a198cff2054f"
meta-st-x-linux-ai   = "<unknown>:<unknown>"
meta-rauc            = "HEAD:b27b59e2f9a3a7446f740b10560462076ef07009"
meta-st-openstlinux-phytec = "HEAD:52620f3fbe5171b121767a6860b4110cef1f1f1d"
meta-python          
meta-oe              
meta-gnome           
meta-initramfs       
meta-multimedia      
meta-networking      
meta-webserver       
meta-filesystems     
meta-perl            = "HEAD:402affcc073db39f782c1ebfd718edd5f11eed4c"
meta-st-stm32mp      = "HEAD:b7accec80a2903c75d4129cf2b3bf6e15f5410c0"
meta-qt5             = "HEAD:ae8a97f79364bed1abc297636f7933d0e35f22be"
meta-st-openstlinux  = "HEAD:613b158e6ee5b575e7de1251eae389c72e376043"
meta                 = "HEAD:ab2649ef6c83f0ae7cac554a72e6bea4dcda0e99"

Initialising tasks: 100% |################################################################################################################| Time: 0:00:05
Sstate summary: Wanted 410 Local 347 Mirrors 0 Missed 63 Current 2718 (84% match, 97% complete)
Removing 6 stale sstate objects for arch phycore_stm32mp1_3: 100% |#######################################################################| Time: 0:00:00
NOTE: Executing Tasks
ERROR: onnxruntime-1.11.0+gitAUTOINC+b713855a98-r0 do_configure: ExecutionError('/home/lupo/PhytechSTM/build-openstlinuxweston-phycore-stm32mp1-3/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/onnxruntime/1.11.0+gitAUTOINC+b713855a98-r0/temp/run.do_configure.318202', 1, None, None)
ERROR: Logfile of failure stored in: /home/lupo/PhytechSTM/build-openstlinuxweston-phycore-stm32mp1-3/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/onnxruntime/1.11.0+gitAUTOINC+b713855a98-r0/temp/log.do_configure.318202
Log data follows:
| DEBUG: Executing python function extend_recipe_sysroot
| NOTE: Direct dependencies are ['/home/lupo/PhytechSTM/layers/meta-openembedded/meta-python/recipes-devtools/python/python3-pybind11_2.8.1.bb:do_populate_sysroot', '/home/lupo/PhytechSTM/layers/openembedded-core/meta/recipes-core/glibc/glibc_2.35.bb:do_populate_sysroot', '/home/lupo/PhytechSTM/layers/openembedded-core/meta/recipes-devtools/cmake/cmake-native_3.22.3.bb:do_populate_sysroot', '/home/lupo/PhytechSTM/layers/openembedded-core/meta/recipes-devtools/gcc/gcc-cross_11.4.bb:do_populate_sysroot', '/home/lupo/PhytechSTM/layers/openembedded-core/meta/recipes-devtools/gcc/gcc-runtime_11.4.bb:do_populate_sysroot', '/home/lupo/PhytechSTM/layers/openembedded-core/meta/recipes-devtools/python/python3_3.10.14.bb:do_populate_sysroot', '/home/lupo/PhytechSTM/layers/openembedded-core/meta/recipes-devtools/quilt/quilt-native_0.67.bb:do_populate_sysroot', 'virtual:native:/home/lupo/PhytechSTM/layers/openembedded-core/meta/recipes-devtools/ninja/ninja_1.10.2.bb:do_populate_sysroot', 'virtual:native:/home/lupo/PhytechSTM/layers/openembedded-core/meta/recipes-devtools/patch/patch_2.7.6.bb:do_populate_sysroot', 'virtual:native:/home/lupo/PhytechSTM/layers/openembedded-core/meta/recipes-devtools/pseudo/pseudo_git.bb:do_populate_sysroot', 'virtual:native:/home/lupo/PhytechSTM/layers/openembedded-core/meta/recipes-devtools/python/python3-numpy_1.22.3.bb:do_populate_sysroot', 'virtual:native:/home/lupo/PhytechSTM/layers/openembedded-core/meta/recipes-devtools/python/python3_3.10.14.bb:do_populate_sysroot', 'virtual:native:/home/lupo/PhytechSTM/layers/openembedded-core/meta/recipes-extended/unzip/unzip_6.0.bb:do_populate_sysroot']
| NOTE: Installed into sysroot: []
| NOTE: Skipping as already exists in sysroot: ['python3-pybind11', 'glibc', 'cmake-native', 'gcc-cross-arm', 'gcc-runtime', 'python3', 'quilt-native', 'ninja-native', 'patch-native', 'pseudo-native', 'python3-numpy-native', 'python3-native', 'unzip-native', 'texinfo-dummy-native', 'xz-native', 'flex-native', 'libtool-native', 'mpfr-native', 'zstd-native', 'linux-libc-headers', 'libmpc-native', 'zlib-native', 'binutils-cross-arm', 'gmp-native', 'gnu-config-native', 'python3-installer-native', 'python3-wheel-native', 'python3-cython-native', 'python3-setuptools-native', 'readline', 'libnsl2', 'bzip2', 'xz', 'openssl', 'zlib', 'libxcrypt', 'libtirpc', 'ncurses', 'gdbm', 'sqlite3', 'util-linux-libuuid', 'libffi', 'opkg-utils', 're2c-native', 'ncurses-native', 'curl-native', 'bzip2-native', 'boost', 'attr-native', 'libtirpc-native', 'openssl-native', 'sqlite3-native', 'libnsl2-native', 'libffi-native', 'util-linux-libuuid-native', 'readline-native', 'gdbm-native', 'libgcc', 'gettext-minimal-native', 'm4-native', 'python3-flit-core-native', 'cryptodev-linux', 'icu', 'perl-native', 'autoconf-archive', 'make-native']
| DEBUG: Python function extend_recipe_sysroot finished
| DEBUG: Executing shell function do_configure
| -- The C compiler identification is GNU 11.4.0
| -- The CXX compiler identification is GNU 11.4.0
| -- Detecting C compiler ABI info
| -- Detecting C compiler ABI info - done
| -- Check for working C compiler: /home/lupo/PhytechSTM/build-openstlinuxweston-phycore-stm32mp1-3/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/onnxruntime/1.11.0+gitAUTOINC+b713855a98-r0/recipe-sysroot-native/usr/bin/arm-ostl-linux-gnueabi/arm-ostl-linux-gnueabi-gcc - skipped
| -- Detecting C compile features
| -- Detecting C compile features - done
| -- Detecting CXX compiler ABI info
| -- Detecting CXX compiler ABI info - done
| -- Check for working CXX compiler: /home/lupo/PhytechSTM/build-openstlinuxweston-phycore-stm32mp1-3/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/onnxruntime/1.11.0+gitAUTOINC+b713855a98-r0/recipe-sysroot-native/usr/bin/arm-ostl-linux-gnueabi/arm-ostl-linux-gnueabi-g++ - skipped
| -- Detecting CXX compile features
| -- Detecting CXX compile features - done
| Building ONNX Runtime for armv7ve
| -- Performing Test onnxruntime_HAVE_BUILTIN_ATOMICS
| -- Performing Test onnxruntime_HAVE_BUILTIN_ATOMICS - Success
| -- Looking for pthread.h
| -- Looking for pthread.h - found
| -- Performing Test CMAKE_HAVE_LIBC_PTHREAD
| -- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
[... 213 lines cut ...]
| 
| 
| CMake Generate step failed.  Build files cannot be regenerated correctly.
| WARNING: exit code 1 from a shell command.
ERROR: Task (/home/lupo/PhytechSTM/layers/meta-st-x-linux-ai/recipes-frameworks/onnxruntime/onnxruntime_git.bb:do_configure) failed with exit code '1'
NOTE: Tasks Summary: Attempted 7528 tasks of which 7504 didn't need to be rerun and 1 failed.
NOTE: Writing buildhistory
NOTE: Writing buildhistory took: 4 seconds

Summary: 1 task failed:
  /home/lupo/PhytechSTM/layers/meta-st-x-linux-ai/recipes-frameworks/onnxruntime/onnxruntime_git.bb:do_configure
Summary: There was 1 ERROR message, returning a non-zero exit code.

-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

My environnement : 
Ubuntu 22.04
PHYTEC STM32MP157 
Yocto Project 4.0.19 (2022-04.19-kirkstone)
BSP PHYTEC 

When i add IMAGE_INSTALL:append = " onnxruntime", i have this error 
numpy is well installed on this yocto project

Can you help me please ?

mguSTM, 2024-11-19:
Hello @IlLupo12725 

The issue that you are facing, is a known issue that we fixed in x-linux-ai v5.0.0.
As this version is not compatible with your yocto kirkstone version, you will need to add the following line in the onnxruntime_git.bb recipe : 

DEPENDS = "\
	${PYTHON_PN}-numpy-native \
	**${PYTHON_PN}-numpy \**
	${PYTHON_PN}-pybind11 \
	${PYTHON_PN}-native \
	${PYTHON_PN} \

It should add the missing numpy dependency for onnxruntime compilation.

Best regards,

Maxence

vinceab, 2025-01-02:
Answered.

closed 2025-01-02 by vinceab